usb4java/usb4java #43: Native library not found in classpath: /org/usb4java/linux-arm/libusb4java.so or Raspberry PI

hofline, 2015-01-09:
Hello, I´m trying to run a simple java program for discover usb devices but I´m getting this error

Exception in thread "AWT-EventQueue-0" org.usb4java.LoaderException: Native library not found in classpath: /org/usb4java/linux-arm/libusb4java.so
        at org.usb4java.Loader.extractLibrary(Loader.java:281)
        at org.usb4java.Loader.load(Loader.java:358)
        at org.usb4java.javax.Services.<init>(Services.java:59)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:408)
        at java.lang.Class.newInstance(Class.java:433)
        at javax.usb.UsbHostManager.createUsbServices(Unknown Source)
        at javax.usb.UsbHostManager.getUsbServices(Unknown Source)

All usb4java reference are in lib folder and classpath is pointing to it.

Linux raspberrypi 3.12.35+ #730 PREEMPT Fri Dec 19 18:31:24 GMT 2014 armv6l GNU/Linux

Thanks

geared554, 2015-08-02:
Same thing happens to me. Have you solved this issue?

hofline, 2015-08-03:
Yes, was a udev configuration issue but I can´t find my configuration backup. Later I will get it from my raspberry pi if you wish

geared554, 2015-08-04:
Thanks for reply, but there's no need. I figured it out.

allengregoryiv, 2015-12-26:
Can one of you post the solution you found, I may be running into a similar problem