usb4java/usb4java #45: usb4java 1.2.0 crashes on OS X 10.10.2 Yosemite

kreutter, 2015-02-06:
The following code fragment from one of the examples crashes the JVM:

```java
DeviceList list = new DeviceList();
int result = LibUsb.getDeviceList(null, list);
```

Crash log:
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007fff8f7335c1, pid=26540, tid=4867
#
# JRE version: Java(TM) SE Runtime Environment (7.0_71-b14) (build 1.7.0_71-b14)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (24.71-b01 mixed mode bsd-amd64 compressed oops)
# Problematic frame:
# C  [libsystem_pthread.dylib+0x15c1]  _pthread_mutex_lock+0x17

[...]

Stack: [0x0000000110661000,0x0000000110761000],  sp=0x000000011075ffc0,  free space=1019k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [libsystem_pthread.dylib+0x15c1]  _pthread_mutex_lock+0x17
C  [libusb4java.dylib+0x73b9]  libusb_get_device_list+0xa9
C  [libusb4java.dylib+0x1cae]  Java_org_usb4java_LibUsb_getDeviceList+0xae
j  org.usb4java.LibUsb.getDeviceList(Lorg/usb4java/Context;Lorg/usb4java/DeviceList;)I+0

[...]

```

malsyned, 2015-03-21:
Have you called LibUsb.init(null) before those lines of code? I was getting segfaults until I did that.

Ingli, 2015-11-10:
Thanks! This does the job also on Ubuntu.

kayahr, 2018-10-08:
This segfault also happens when doing the same with libusb in C directly. I've fixed this now in usb4java 1.3.0-SNAPSHOT by throwing an IllegalStateException when the default context is not initialized.

closed 2018-10-08 by kayahr