usb4java/usb4java #47: Native library not found exception for usb4java usage in android device

bzkrtmurat, 2015-12-03:
I want to develop usb application with usb4java for Android device. I'm using Android Studio as an IDE and Ubuntu 14.04 as an operating system. I have added usb4java-1.2.0 file to my project. I can use Libusb function in my application. Firstly I wrote LibUsb.init() function in my MainActivity but there is an error about native library.  I'm getting classloader exception. 

Caused by: org.usb4java.LoaderException: Native library not found in classpath: /org/usb4java/linux-armv7l/libusb4java.so

I have analyzed Unsupported Platform tag in http://usb4java.org/nativelibs.html website. But I couldn't solve my problem with this way. After that I read https://github.com/usb4java/libusb4java and again I couldn't solve my problem with your way. Can you help me please.

koh-osug, 2016-02-14:
Not sure what you have tried. But when you have followed the steps, you should have build a native library for Android and placed them into `/org/usb4java/linux-armv7l/libusb4java.so`. And you are still getting the error?

bzkrtmurat, 2016-02-15:
I'm still getting this error. I couldn't find solution so now I'm using android usb host api. But there's is different problem for asynchronous bulk transfer in usb host api. If you could help me I would start to use usb4java again. Thank you for your reply

kayahr, 2018-10-10:
usb4java 1.3.0-SNAPSHOT now supports "armv7l" architecture name and maps it to the "arm" binary.

But I'm not sure if this helps. Never tried using usb4java on Android. Finding the right binary is easy (and should work now with 1.3.0-SNAPSHOT) but I can't test if the binary really works on Android.

closed 2018-10-10 by kayahr