usb4java/usb4java #49: Failed to load library on ARM (java.lang.UnsatisfiedLinkError)

allengregoryiv, 2015-12-26:
I'm trying to use usb4java to make an android accessory from a robot controller with an ARM process running a version of Angstrom. I'm currently not able to get the library to load properly.

```
ERROR Unhandled exception: java.lang.UnsatisfiedLinkError: 
/var/volatile/tmp/usb4java4482639387304626765.tmp/libusb4java.so: 
ld-linux-armhf.so.3: cannot open shared object file: No such file or directory at 
[java.lang.ClassLoader$NativeLibrary.load(Native Method), 
java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1929), 
java.lang.ClassLoader.loadLibrary(ClassLoader.java:1814), 
java.lang.Runtime.load0(Runtime.java:809), java.lang.System.load(System.java:1083), 
org.usb4java.Loader.load(Loader.java:358), org.usb4java.LibUsb.<clinit>(LibUsb.java:640), 
org.spectrum3847.lib.drivers.AndroidAccessory.init(AndroidAccessory.java:79), 
```

thank you for providing a great library and your help,
Allen

java-dog, 2016-01-06:
I've tried mostly the same and ran into the same problem. It seems you have a linux kernel with soft-floating point running on your ARM. Your kernel uses ld-linux.so.3 and does not supply ld-linux-armhf.so.3 (hardware float). After changing my toolchain to hardware floating point and installing the other JVM, it worked.

But you have to compile libusb4java anyhow and can not use the precompiled binary from the website, because it probably won't run on your ARM machine very well. I found out, that the precompiled binary does not react on hotplug events, because it was compiled from a different source.
If you have to compile it anyhow, you can probably link it with the right libusb and ld-linux.so of your kernel and won't get the above error.

kayahr, 2018-10-10:
I'm closing this because I think java-dog correctly explained to problem already. usb4java does not provide a prebuilt binary for armel. 

It would be possible to add support for it, but unfortunately I have no idea how to determine in Java if system is armhf or armel. This would be necessary to decide which library to load. So if someone knows how this works and is willing to help with testing I could add armel support.

closed 2018-10-10 by kayahr