usb4java/usb4java #51: Can't claim device on Mac

tresf, 2016-02-28:
After a few days with usb4java, I've successfully:

 * Disable System Integrity protection on OSX 10.11 "El Capitan" via [stackexchange/a/208481](http://apple.stackexchange.com/a/208481) so that non-signed kernel extensions can be used.
 * Implemented an OS X `kext` kernel extension to prevent kernel driver from claiming device by `vendorId` and `productId` per [libusb/wiki/FAQ](https://github.com/libusb/libusb/wiki/FAQ#how-can-i-run-libusb-applications-under-mac-os-x-if-there-is-already-a-kernel-extension-installed-for-the-device)

... but the device won't list as "active".  What I mean is... `usbDevice.getActiveUsbConfiguration()` returns `null`.

So my next step was to grab the first `UsbConfiguration` available.  I did this by:

```java
for (Object o : usbDevice.getUsbConfigurations()) {
   if (o != null) return (UsbConfiguration)o;
}
```

Which allows me to successfully do things like `usbInterface.getUsbInterfaceDescriptor()`,  `usbEndpoint.getUsbEndpointDescriptor()`.

However, when I try to claim the device, I get the following exception:

```
Configuration is not active
```

This leads me down some paths of using the Low-Level API to set the active alternate configuration, but I'm starting to get lost.  Is there a simple way, using the High-Level API to set a particular configuration as active?  Crawling the internet suggests explicitely setting the active configuration [may not be uncommon](http://libusb.6.n5.nabble.com/Mac-OS-X-configuration-not-set-td4384677.html).  Any help appreciated. :+1:

kayahr, 2018-10-21:
usb4java 1.3.0 was released today which now uses libusb 1.0.22 which may have fixed this issue. Please try again with this new version and give feedback here if the exact same problem still occurs. If it still doesn't work but it is not the exact same result then please open a new issue.

closed 2018-10-21 by kayahr

tresf, 2018-10-22:
@kayahr thanks for the update.  I've moved away from `usb4java` for HID devices due to the `kext` complexities above, (and we don't have much demand for non-HID hardware currently) so it'll be some time before this is re-tested on macOS against non-HID hardware. I'll let you know when that time comes.