usb4java/usb4java #53: Devicelist returned from LibUsb.getDeviceList not actualized

tondmn, 2016-10-04:
Hi

In my setup, Linux Centos 6, Java version 1.8.0_102-b14 I noticed the following:

When an USB device is added or removed from the bus, a followed LibUsb.GetDeviceList is not updated with this event. The same number of devices is returned.

I tested it with the native libusb-1.0.20, and there seems to work correctly.

After doing a new LibUsb.init, the list is indeed actual again, but this is not what you expect, since the native function does it well.

[out.pdf](https://github.com/usb4java/usb4java/files/508611/out.pdf)

Regards
Ton

vdegrandpre, 2016-10-07:
Same thing here, on Raspbian (Raspberry Pi 3). Inserting USB get detected event go fine, removing it get detected fine also but re-reconnecting fails to detect anything.

kayahr, 2018-10-21:
This doesn't sound like anything which may have been repaired in libusb 1.0.22 because you say it was working for you with native libusb 1.0.20. But I was not able to reproduce it with the new usb4java 1.3.0 which now uses libusb 1.0.22. I used your test program, it displayed 13 devices, I unplugged one device and pressed enter, it displayed 12 devices, I plugged the device back in, pressed enter and it displayed 13 devices again. I guess this is what you expected?

Well, maybe it was accidentally fixed in usb4java. I'm closing this for now. If you can still reproduce this behavior with usb4java 1.3.0 then please open a new issue and also provide the C source code which was working for you. 

And it would be easier if you embed the source code next time properly (markdown syntax) in the issue description instead of attaching a PDF. :)

closed 2018-10-21 by kayahr