usb4java/usb4java #55: Application crashes with SEGV when plugging or unplugging USB device

RainerKlute, 2016-11-30:
I have a rather complex application which runs flawlessly on openSUSE 13.2 with a 3.16.7-48 Linux kernel, but crashes on Ubutu 16.04 LTS with a 4.4.0-31 Linux kernel.

The crash happens reliably and reproducibly each time a USB device gets plugged or unplugged and *before* my **UsbServiceListener** gets called. Since the application is multithreaded, I suspect some threading issues. In comparison, a very simple single-threaded sample application that just listens for attach and detach events and prints some device properties runs nicely on both systems.

I attached *gdb* to processes of my complex application and got similar backtracks, seemingly pointing to *udev_device_unref ()* in *libusb*.

Attaching a USB device:
```
(gdb) continue 
Continuing.
[New Thread 0x7fcab04bf700 (LWP 27966)]

Thread 29 "java" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fcab2bed700 (LWP 27896)]
__GI___libc_free (mem=0x1b) at malloc.c:2952
2952    malloc.c: Datei oder Verzeichnis nicht gefunden.
(gdb) backtrace 
#0  __GI___libc_free (mem=0x1b) at malloc.c:2952
#1  0x00007fcab2c12c79 in udev_device_unref () from /tmp/usb4java5791198224965343141.tmp/libusb4java.so
#2  0x00007fcab2c12c5e in udev_device_unref () from /tmp/usb4java5791198224965343141.tmp/libusb4java.so
#3  0x00007fcab2c10be6 in ?? () from /tmp/usb4java5791198224965343141.tmp/libusb4java.so
#4  0x00007fcab2c10db8 in ?? () from /tmp/usb4java5791198224965343141.tmp/libusb4java.so
#5  0x00007fcb4a0db6fa in start_thread (arg=0x7fcab2bed700) at pthread_create.c:333
#6  0x00007fcb4a815b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
(gdb) 
```

Detaching a USB device:
```
(gdb) continue 
Continuing.

Thread 28 "java" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7f3fd0c9b700 (LWP 27605)]
__GI___libc_free (mem=0x1) at malloc.c:2952
2952    malloc.c: Datei oder Verzeichnis nicht gefunden.
(gdb) backtrace 
#0  __GI___libc_free (mem=0x1) at malloc.c:2952
#1  0x00007f3fd0cc0c67 in udev_device_unref () from /tmp/usb4java4778463921093166841.tmp/libusb4java.so
#2  0x00007f3fd0cbebe6 in ?? () from /tmp/usb4java4778463921093166841.tmp/libusb4java.so
#3  0x00007f3fd0cbedb8 in ?? () from /tmp/usb4java4778463921093166841.tmp/libusb4java.so
#4  0x00007f4053fb06fa in start_thread (arg=0x7f3fd0c9b700) at pthread_create.c:333
#5  0x00007f40546eab5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
(gdb) 
```

maciekd, 2017-08-09:
Any news on this? I'm getting the same problem.

kayahr, 2018-10-21:
usb4java 1.3.0 was released today which now uses libusb 1.0.22 which may have fixed this issue. Please try again with this new version and give feedback here if the exact same problem still occurs. If it still doesn't work but it is not the exact same result then please open a new issue.

closed 2018-10-21 by kayahr