usb4java/usb4java #57: usb4java for aarch64?

chudgoo, 2017-01-31:
Is anyone aware of a version of libusb4java that has been compiled for use on AArch64?

Specifically the Amlogic S905 in the ODROID C2 running either 64-bit Ubuntu Mate 16.04 or DietPi?

kayahr, 2018-10-08:
Current Snapshot Version 1.3.0-SNAPSHOT now has aarch64 (arm64) support. It is compiled on Debian Stretch on a aarch64 qemu VM. I hope this works on your devices, can't really test it myself because I have no real aarch64 device. So I would be happy to get some feedback.

https://nexus.ailis.de/content/groups/public/org/usb4java/usb4java/1.3.0-SNAPSHOT/

kayahr, 2018-10-21:
1.3.0 with aarch64 support was released today. I hope it works for you.

closed 2018-10-21 by kayahr