usb4java/usb4java #59: USB error 5: Unable to read config descriptor: Entity not found

Tsyklop, 2017-06-05:
```
Exception in thread "main" org.usb4java.LibUsbException: USB error 5: Unable to read config descriptor: Entity not found
    at com.tsyklop.usb.Usb.dumpConfigurationDescriptors(Usb.java:197)
    at com.tsyklop.usb.Usb.dumpDevice(Usb.java:254)
    at com.tsyklop.usb.Usb.dumpDevicesList(Usb.java:158)
    at com.tsyklop.Main.main(Main.java:79)
```

code:

```
public void dumpDevicesList() {
        try {
            for (Device device: list) {
                System.out.println("\n=======================\n");
                dumpDevice(device);
                System.out.println("\n=======================\n");
            }
        } finally {
            LibUsb.freeDeviceList(list, true);
        }
    }

    private void dumpConfigurationDescriptors(final Device device, final int numConfigurations) {
        for (byte i = 0; i < numConfigurations; i += 1) {
            final ConfigDescriptor descriptor = new ConfigDescriptor();
            final int result = LibUsb.getConfigDescriptor(device, i, descriptor);
            if (result < 0) {
                throw new LibUsbException("Unable to read config descriptor", result);
            }
            try {
                System.out.println(descriptor.dump().replaceAll("(?m)^", "  "));
            } finally {
                // Ensure that the config descriptor is freed
                LibUsb.freeConfigDescriptor(descriptor);
            }
        }
    }

    private void dumpDevice(final Device device) {
        // Dump device address and bus number
        final int address = LibUsb.getDeviceAddress(device);
        final int busNumber = LibUsb.getBusNumber(device);

        System.out.println(String.format("Device %03d/%03d", busNumber, address));

        // Dump port number if available
        final int portNumber = LibUsb.getPortNumber(device);
        if (portNumber != 0) {
            System.out.println("Connected to port: " + portNumber);
        }

        // Dump parent device if available
        final Device parent = LibUsb.getParent(device);
        if (parent != null) {
            final int parentAddress = LibUsb.getDeviceAddress(parent);
            final int parentBusNumber = LibUsb.getBusNumber(parent);
            System.out.println(String.format("Parent: %03d/%03d", parentBusNumber, parentAddress));
        }

        // Dump the device speed
        System.out.println("Speed: " + DescriptorUtils.getSpeedName(LibUsb.getDeviceSpeed(device)));

        // Read the device descriptor
        final DeviceDescriptor descriptor = new DeviceDescriptor();
        int result = LibUsb.getDeviceDescriptor(device, descriptor);
        if (result < 0) {
            throw new LibUsbException("Unable to read device descriptor", result);
        }

        // Try to open the device. This may fail because user has no
        // permission to communicate with the device. This is not
        // important for the dumps, we are just not able to resolve string
        // descriptor numbers to strings in the descriptor dumps.
        DeviceHandle handle = new DeviceHandle();
        result = LibUsb.open(device, handle);
        if (result < 0) {
            System.out.println(String.format("Unable to open device: %s. " + "Continuing without device handle.", LibUsb.strError(result)));
            handle = null;
        }

        // Dump the device descriptor
        System.out.print(descriptor.dump(handle));

        // Dump all configuration descriptors
        dumpConfigurationDescriptors(device, descriptor.bNumConfigurations());

        // Close the device if it was opened
        if (handle != null) {
            LibUsb.close(handle);
        }
    }
```

kayahr, 2018-10-21:
usb4java 1.3.0 was released today which now uses libusb 1.0.22 which may have fixed this issue. Please try again with this new version and give feedback here if the exact same problem still occurs. If it still doesn't work but it is not the exact same result then please open a new issue.

closed 2018-10-21 by kayahr

gegomu, 2019-03-27:
It still happends