usb4java/usb4java #60: Status of this project

MichaelJAllen, 2017-12-15:
Firstly, I just want to say how grateful am I to the developers of this excellent library! Thanks, guys!

However, I'm a little nervous about its status. The last issue to be closed was on Feb 19, 2016 (almost 2 years ago), the last commit on Aug 5 2015 (well over 2 years ago), and the last PR to be merged on Mar 2, 2014 (almost 4 years ago). Looking through the open issues, there don't even appear to have been any comments from a developer in a long, long time. The web-site also fails to reflect the change in the project's licensing. It's looking very much as though this project is - at best - lacking support, and - at worst - completely dead.

I appreciate that maintaining the library can be a lot of effort, particularly when you're volunteering your time and effort. However, I'd welcome a statement of intent regarding the future of this library. What's the roadmap for future releases? If high-quality PRs are submitted, will they be accepted? Will a new maintainer be sought if necessary?

sickworm, 2018-01-17:
It's a wired thing that all the USB driver libraries on Java platform have stopped development for years... Am I on the wrong way of controlling USB devices by Java?

sickworm, 2018-01-17:
Here are some libraries I know: usb4java, jsr080, jUSB

kayahr, 2018-01-17:
Late answer for the first question: Yes, usb4java is no longer under active development. Reason for this is that in my current job I no longer use Java and currently don't work on USB-related projects. So there is currently no motivation (and no time) to work on usb4java.

I started this project years ago because of the frustration that there is no working USB library for Java. And now it looks like I just added another library to the pool of dead projects. I'm sorry for this. But at least usb4java still compiles and most likely works in general at least on Debian Jessie and Windows 10 (Both are still tested automatically every night). OSX and Raspberry tests no longer work because lack of test hardware.

If high-quality PRs are submitted then I try to process them. Up to now there was only one such PR and the author of it became the co-author of usb4java. So maybe this happens again in the future ;-)

For the second question (If Java is the right choice for USB): My personal opinion is that Java is not the ideal platform for USB related stuff. Today I would choose C++ because libusb can be used directly without any badly maintained wrapper library and with some discipline it is easy enough to write cross-platform applications with C++.

MichaelJAllen, 2018-01-17:
Klaus, thanks for your reply! I appreciate your efforts creating and maintaining this library! Thanks!

Thanks, also, for your honesty regarding its status. Alas, I don't have the option of using _C++_ (I actually use _Scala_ - another _JVM_ language - instead of _Java_). Personally, I think it's a big mistake that _Oracle_ (and _Sun_ before them) haven't seen fit to incorporate _USB_ support into the main _Java_ runtime libraries. C'est la Vie...

llongi, 2018-01-17:
Same here, the focus at my job is now on a C/C++ version of our software, and the Java one is somewhat in a maintainance mode.
Also, fundamentally usb4java just works, the Java software we use is used daily on Linux, OSX and Win 7-10, and working with the devices just works.
It might be an idea to just release a new usb4java that uses the latest libusb-1.0.22 as soon as it's released (RC phase currently), so that any bugs there are fixed.
It's not like there have been many new features in libusb, the API has seen minimal improvements and none that are vital.

llongi, 2018-01-17:
At least the low level part, libusb4java, is pretty rock-solid (modulo libusb bugs).
I've never used the higher-level javax.usb abstraction, where almost all of the tickets/queries are.
There's also a bunch related to ARM/Raspberry, but fixing that is not easy usually, you'd have to compile the libusb shared library for all the subtly different architectures there, and provide a means to identify and load the right one.
Just as a reminder, this changed in the libusb low-level API since we last did a release:
- added LIBUSB_SPEED_SUPER_PLUS = 5 to libusb_speed enum
- new functions libusb_dev_mem_alloc() and libusb_dev_mem_free() (only useful on Linux for now)
- new function libusb_interrupt_event_handler()
- new function libusb_free_pollfds()
- new enum libusb_option with values LIBUSB_OPTION_LOG_LEVEL, LIBUSB_OPTION_USE_USBDK and corresponding function libusb_set_option()
So fundamentally if you ignore the Linux only dev_mem support and the Windows only LIBUSB_OPTION_USE_USBDK, nothing new of critical importance.

kayahr, 2018-10-12:
I found some time in the last few days to work on usb4java and bumped libusb to the latest 1.0.22 version and implemented all the stuff Luca listed in the last comment and additionally also the libusb_get_pollfds() function which was also missing.

I want to do a few more tests (Especially the new USBDK stuff on Windows) and then I guess usb4java is ready for a 1.3.0 release.

I'm closing this issue because it is not a bug report.

closed 2018-10-12 by kayahr