usb4java/usb4java #68: macOS High Sierra can't claim interface (returned: -99)

developersu, 2019-03-13:
**Environment:**
 - OS: MacOS High Sierra 10.12.4
 - Java version: Oracle 1.8.0_201 (Oracle 11 also)
 - usb4java version: 1.3.0

Good day!
I use in my project usb4java 1.3.0 (from maven) and it works as expected on MacOS Mojave, Windows 7-to-10, Linux. But on macOS 'High Sierra' it returns error on `LibUsb.claimInterface(...)`.
Returned value: `-99 (ERROR_OTHER)`

No issues if I switch to version 1.2.0. It works as expected.

I really like 1.3.0, so could you fix latest release please?

Thanks a lot!

llongi, 2019-03-13:
There are known problems with libusb 1.0.22 on MacOS X, and that's the libusb version used in usb4java 1.3.0. When libusb 1.0.23 is going to be released with the fixes we'll of course also update usb4java.
See: https://github.com/libusb/libusb/issues/519

TMaxElectronics, 2021-04-05:
Is this project still active? libusb 1.0.23 and 1.0.23 has been released but no news here ☹️