usb4java/usb4java #72: Read from Rfid reader and get USB error 1: Transfer error on interrupt endpoint: Input/Output Error

tyrion9, 2019-06-29:
**Environment:**
 - OS: Windows 10 64bit
 - Java version Oracle Java 8
 - usb4java version 1.3.0

**Bug description**
I want to read data from Rfid Reader, all procedures is ok, but when `pipe.syncSubmit(buffer);.` I catched an error: javax.usb.UsbPlatformException: USB error 1: Transfer error on interrupt endpoint: Input/Output Error

**Reproduction**
``    public static void main(String[] args) throws UsbException {
        UsbHub hub = UsbHostManager.getUsbServices().getRootUsbHub();
        UsbDevice selectedUsb = null;
        UsbInterface iFace = null;

        for (UsbDevice device : (List<UsbDevice>) hub.getAttachedUsbDevices()) {
            UsbDeviceDescriptor desc = device.getUsbDeviceDescriptor();
            System.out.println(desc.idProduct() + "." + desc.idVendor());
            if (desc.idProduct() == idProduct && desc.idVendor() == idVendor)
                selectedUsb = device;
        }

        UsbConfiguration configuration = selectedUsb.getActiveUsbConfiguration();
        System.out.println(selectedUsb.getUsbDeviceDescriptor());

        final UsbPipe pipe;
        UsbPipe pipeTmp = null;
        int length = 0;

        for (Object i : selectedUsb.getActiveUsbConfiguration().getUsbInterfaces()) {
            iFace = (UsbInterface) i;
            iFace.claim(new UsbInterfacePolicy()
            {
                @Override
                public boolean forceClaim(UsbInterface usbInterface)
                {
                    return true;
                }
            });
            for (Object e : iFace.getUsbEndpoints()) {
                UsbEndpoint endp = (UsbEndpoint) e;
                if (endp.getDirection() == UsbConst.ENDPOINT_DIRECTION_IN) {
                    pipeTmp = endp.getUsbPipe();
                    break;
                }
            }
        }

        pipe = pipeTmp;
        pipe.open();

        byte[] buffer = new byte[8];
        pipe.syncSubmit(buffer);
    }
`.`

tyrion9, 2019-06-30:
It seems my rfid reader (http://www.sycreader.com/index.php/en/walkthroughmetaldetector/317.html) device is not working with libusb driver. I install Zadig to replace with libusb driver and exam the device that is not working properly. 

usb4java uses native libusb 1.0, I think it can't work.

bbrand84, 2019-09-13:
I got that error when I set up my byte buffer too small. To fix this issue, I set it to the maximum size that I got from the endpoint description of the reading endpoint (ENDPOINT_DIRECTION_IN, 0x81), instead of 4 bytes (8 in your example) that I had the impression would be enough to contain the answer. Hope that helps some people.