usb4java/usb4java #85: Attached and Detached events are not received for the device that has lengthy serial number

sibiarunachalam, 2021-05-10:
**Environment:**
 - OS: Windows 10
 - Java version: Oracle Java 8
 - usb4java version: 1.3.0

**Bug description**
We use usb4java in our application to detect attached and detached events
It works well except the scenario we have identified recently.
Usb4java does not detect the ATTACHED event for the device that has serial number with more than 100 characters when it is plugged in.

Vendor ID: 0781
Product ID: 5591
Serial No: 0101c14ea1b1ea6e90aa9860523346cf654a93d436044650cdf7c88765d187e9e52300000000000000000000fc1c000e000f2500915581072a2ad2ae

**Reproduction**
Simply plugin the device that has serial number with more than 100 characters

**Expected behavior**
Attached and Detached events should be received when plug and unplug the device

sibiarunachalam, 2021-06-15:
Events are detected as expected in Windows when listdevs.exe from libusb is executed
But it doesn't work when ListDevices.java from usb4jave is used

sibiarunachalam, 2021-06-16:
Any updates guys?
It seems a bug in either usb4java or libusb4java

sibiarunachalam, 2021-06-16:
Thanks guys.
I've identified the problem.
I could recreate the problem in libusb-1.0.22, but not in 1.0.24.
So there is no issues with usb4java projects.
usb4java needs to be updated to libusb-1.0.24.

sibiarunachalam, 2021-06-16:
So next question is when will you release new version of usb4java with this fix (upgrading libusb to 1.0.24 or latest)?

stodge, 2025-05-23:
I'm guessing never. No release since 2018.